sourcebots/sbot #1: Servo Assembly Support

trickeydan, 2019-05-19:
Blocked on upstream: https://github.com/j5api/j5/issues/66

PeterJCLaw, 2019-06-09:
Ah, is the intention that we merge the user-API for both the Arduino and Servo Board even though they're separate boards (with the move to use SR v4 Servo Boards)?

I think that that would be very confusing for competitors and we should change the API so they're separate. (I created #4 on that assumption before realising what this ticket was).

trickeydan, 2019-06-09:
This was specifically for the SB 'Servo Assembly', aka. an Arduino with a servo hat.

We aren't shipping the servo hats this year, in favour of the SR Servo boards.

The Servo board has a separate API to the Arduino for obvious reasons.

closed 2019-07-27 by trickeydan