sourcebots/sbot #18: Update all pictures in docs.

trickeydan, 2019-07-15:
`arduino.png` is a picture of legacy servo assembly. I suggest deleting it to prevent future confusion, and renaming `arduino_headers.png` to `arduino.png`.

_Originally posted by @kierdavis in https://github.com/sourcebots/sbot/timeline_

kierdavis, 2019-07-15:
Also from #11:

> @kierdavis: There is both a `pi.jpg`, used in `docs/source/kit/index.rst`, and a `pi2.jpg`, used here. Do we really need both?

> @trickeydan `pi2.jpg` is a much better image. I suggest removing `pi.jpg`

kierdavis, 2019-07-24:
Also from #11:

> We should really add attribution for photos we used. I'd suggest we even include it for ones we've taken ourselves, so that other people can replicate it if they reuse the photo elsewhere.
> 
> * [ ]  `arduino_headers.png`: SourceBots (assuming this is the photo you took yesterday)
> * [ ]  `arduino_pinout.png`: attribution already included in image
> * [ ]  `arduino.png`: SourceBots (note that one of my other comments suggests deleting this anyway)
> * [ ]  `battery.jpg`: ???
> * [ ]  `mcv4b.png`: Student Robotics
> * [ ]  `mcv4b_diagram.png`: Student Robotics
> * [ ]  `pbv4.png`: Student Robotics
> * [ ]  `pbv4_diagram.png`: Student Robotics
> * [ ]  `pi.jpg`: ???
> * [ ]  `pi2.jpg`: ???
> * [ ]  `sbv4.png`: Student Robotics
> * [ ]  `sbv4_diagram.png`: Student Robotics

closed 2024-05-24 by tpoomlmly