sourcebots/sbot #44: Beep the zone

kierdavis, 2019-08-09:
We could sound a pattern of beeps at startup to indicate the zone number. I suggest inserting this after the beep we currently have that indicates the start button is ready to press, at a different pitch, where the number of beeps corresponds to the zone number (0 to 3).

RealOrangeOne, 2019-08-09:
The fact that we index from 0 (as all things should be) makes this slightly weirder. If they hear no beep, there's no way of knowing if the beep didn't work.

What's the benefit of informing the user which zone they're in? Their zone USB is already the right colour, and they should have been directed to the correct corner.

trickeydan, 2019-08-09:
We think it would be a good debug tool and confirms that the zone USBs have worked. The v3 kit would display the current zone on the screen for example

kierdavis, 2019-08-19:
> The fact that we index from 0 (as all things should be) makes this slightly weirder. If they hear no beep, there's no way of knowing if the beep didn't work.

We could add an additional beep at a different that is always played to indicate competition mode, though this is starting to get quite complicated.

> What's the benefit of informing the user which zone they're in?

As Dan mentioned above it would be good to confirm that the zone USB has worked, especially since I've now observed SourceOS fail to detect a zone USB that was plugged in for a live competition match (that's what the drama around Team 5's robot at Smallpeice 2019 packdown was all about).

Perhaps we don't need to beep to indicate the zone number, just to indicate that an arena USB was detected.

trickeydan, 2019-12-01:
This issue would be fixed by the RGB LED on the Pi Power Hat v2, it could simply flash the appropriate zone colour!

tpoomlmly, 2024-05-24:
Fixed by KCh :)

closed 2024-05-24 by tpoomlmly