sourcebots/sbot #66: Use a less generic name than "metadata.json"

kierdavis, 2019-08-15:
We have observed teams leaving other things on their robot code USB sticks, for example virtualenvs. Some packages installed into a virtualenv also contain a file named "metadata.json". If the sbot metadata search finds one of these files before it finds the one on the arena USB, the wrong file will be loaded (almost certainly not containing "zone" or "is_competition" fields), resulting in bogus MetadataKeyErrors.

We should change the filename to something more specific, e.g. "sbot-metadata.json".

tpoomlmly, 2021-11-13:
At the moment sbot just checks for any `.json` file

trickeydan, 2021-11-13:
If you migrate to Astoria, it looks for 
astoria-metadata.json

tpoomlmly, 2021-11-13:
Good idea; we'll do that instead

closed 2021-11-13 by tpoomlmly